KawaSwitch/Poke-Controller #4: 予期しない上方向入力ga

Moi-poke, 2020-01-07:
##### したこと
timeleap関数を参考に以下のような高速日付変更(乱数消費用)のコード(日付変更がOKからはじまる状態でスタート)の実行・SImple Switch Controllerの操作
##### 結果
あるタイミングで上方向への入力が入りっぱなしになる。(Controllerのhomeボタンで一時的に消えるかも(?))
```
self.press(Button.A, wait=0.2)
self.press(Direction.LEFT)
self.press(Direction.LEFT)
self.press(Direction.LEFT)
self.press(Direction.UP, wait=0.2)  # increment a day
self.press(Direction.RIGHT)
self.press(Direction.RIGHT)
self.press(Direction.RIGHT)
self.press(Button.A, wait=0.2)
```

KawaSwitch, 2020-01-07:
コマンド実行時のJoystick.c側の動作でHATスイッチをニュートラルに入れる動作を忘れていました
[コミット番号 0ab7dda](https://github.com/KawaSwitch/Poke-Controller/commit/0ab7ddad436474c73be2733503decaaeb5e5d58d)にて修正を行いました

こちらの環境では修正を確認いたしました
お手数をおかけしますが念のためそちらの環境でもご確認頂けますでしょうか
なおJoystick.c側の不具合であるため再度makeしていただく必要があります

Moi-poke, 2020-01-08:
数時間ほど動かしてみましたが当該の症状はなく、問題なさそうです。

KawaSwitch, 2020-01-08:
了解いたしました
ご報告いただきありがとうございました！

closed 2020-01-08 by KawaSwitch