KawaSwitch/Poke-Controller #5: LGP Stream Engineを用いた画像認識が行えない

xv13, 2020-01-12:
### ■動作環境
PC : Windows 10 Home(64bit)
キャプボ : AVerMedia AVT-C875

Python : 3.8.1
cv2 : opencv-contrib-python(v4.1.2.30)

### ■問題点
「Poke-Controller」を使用した際に、画像認識が真っ暗で動作しない。
(その際、別のソフトなどでキャプボの画面出力は行っていません。)

### ■確認内容

1. Poke-ControllerでのLGP Stream Engineの動作確認
[image]
CameraID:0(AverMedia C875 Capture) -> 「CameraID0can'topen.」
CameraID:1(LGP Stream Engine) -> 真っ黒な画面が表示される。
CameraID:2-9(割り当て無し) -> 「CameraID0can'topen.」

2. フリーソフトでのLGP Stream Engineの動作確認
[image]
こちら( http://t-ishii.la.coocan.jp/download/AGCRec.html )のフリーソフトにて、
LGP Stream EngineがPC上で正常に動作することの確認を行いました。

キャプボやPythonのバージョンなどに問題があるのでしょうか？
もし解決方法をご存じであれば、ご教授いただければ幸いです。

操作方法や動作要件を満たしていなかったらすみません。

Moi-poke, 2020-01-12:
開発者ではないので見当違いな推測かもしれませんが、LGP Stream Engineの設定で取り込み解像度を720p (1280×720)にすると映るようになるかもしれません。(自分はGV-USB3/HDとアマレコ+ライブ機能の拡張を使用して取り込んでいますが、1080pだと同じ黒画面になり、720pにすると映りました。)

xv13, 2020-01-13:
Moi-poke様
ご回答いただきありがとうございます。

Moi-poke様の仰る通りに取り込み解像度を720pにすることで映像を取り込めるようになりました。
誠にありがとうございます。

**Pokemon Controller動作画面**
[image]

**AVerMedia Stream Engine設定画面**
[image]

closed 2020-01-13 by xv13

KawaSwitch, 2020-01-13:
情報提供ありがとうございます
1080pの取り込み解像度で黒画面となる不具合について把握いたしました
ただこの状況を私の環境で再現できないため現在この問題自体の修正は困難です

不便をおかけして申し訳ありませんが同様の不具合が出てしまう方は, Moi-poke様の方法でのキャプチャをお願いいたします
何かしら分かったことがあったら追記いたします