KawaSwitch/Poke-Controller #7: GUIの起動について

agnnn-ag, 2020-01-17:
こんにちは。GUIをどこから開けるのか分からず、困っています。教えてもらえますか？

KawaSwitch, 2020-01-17:
ご利用いただきありがとうございます。
すみません。まだどこにも書いていませんでした。

[簡易 使用方法]
PokeController/SerialController/Window.py
の実行でGUIが起動します。
python3の環境を整えて必要なパッケージをいくつか導入してから実行する必要があります。

agnnn-ag, 2020-01-17:
ご回答ありがとうございます。コマンドプロンプトが一瞬起動してすぐに消えてしまうのですが、どうしたらいいですか？
tkinter PIL はインストールしました。

KawaSwitch, 2020-01-18:
Windowsにpython環境を構築する際はAnacondaなど導入されていますか
もしくはtkinter, PILはサイトからダウンロードしたインストーラなどを用いてインストールされたということでしょうか
また何かエラーメッセージ等はありますか？

KawaSwitch, 2020-01-18:
tkinter, PIL以外に用いている主なパッケージは次の通りです
- pyserial
- pynput
- pythonnet
- OpenCV

KawaSwitch, 2020-01-18:
丁度新しいPCがあったので以下Windows10+Anacondaで再度試してみました
あくまで一例ですがご参考になれば幸いです

1. Anacondaインストール

2. 下記画像のように"Anaconda Prompt"があるので開く
[image]

3. 次のコマンドをすべて実行し環境を整える
`conda create -n [name] python=3.6`
バージョン3.6でしか動かないパッケージがあるので指定
[name]の部分は好きな仮想環境名に変更
`conda activate [name]`
`pip install opencv-python` (OpenCV)
`pip install pynput` (pynput)
`conda install pyserial` (pyserial)
`conda install Pillow` (PIL)
`conda install -c pythonnet pythonnet` (pythonnet)

4. Poke-Controller/SerialControllerまで移動
`cd [Poke-Controllerまでのパス]\Poke-Controller\SerialController`

5. 実行
`python Window.py`

agnnn-ag, 2020-01-18:
ご丁寧な説明をいただき感謝です。教えていただいて通り行ったところ、起動するのを確認しました。応答なしの状態が続いているのですが、ほかに必要事項はありますでしょうか。

KawaSwitch, 2020-01-18:
私の環境ではこれで起動できましたので必要事項はそろっているはずです
応答なし、というのはどういった状態でしょうか
お手数をかけますが該当箇所のスクリーンショットなどを撮って貼って頂けるとありがたいです

agnnn-ag, 2020-01-18:
[image]
この状態がずっと続いています。よろしくお願いします。

agnnn-ag, 2020-01-18:
タスクマネージャで応答なしの表示が出ています。

KawaSwitch, 2020-01-18:
スクショありがとうございます
v1.0-alphaバージョンをダウンロードしてもらっているみたいですね
一旦git pullして最新にするか, [Poke-Controller ホーム](https://github.com/KawaSwitch/Poke-Controller)の"Clone or Download"ボタン→"Download ZIP"から最新バージョンを落として実行してみて頂けますか

agnnn-ag, 2020-01-18:
[image]
zipを解凍し、anaconda prompt から開いてみたところ、このような表示となりました。重ね重ねの質問となりますが、お教えください。

KawaSwitch, 2020-01-18:
Poke-Controller-masterのフォルダ内にDirectShowLibというフォルダはありますか？

agnnn-ag, 2020-01-18:
あります。
DirectShowLib-2005.dll
lisence.txt
readme.txt
WhatsNew.txt
が入っています。

KawaSwitch, 2020-01-18:
Poke-Controller-master/SerialController/Window.pyの192行目の"/"を"\"に変えて保存し実行できるか試してみてほしいです
Windows環境では"\"は￥マークと表示されるかもしれませんが同じ意味です

agnnn-ag, 2020-01-18:
先ほどの画像と全く同じ表示になりました...

KawaSwitch, 2020-01-18:
.dllにセキュリティがかかっているかもしれません
.dllを右クリック→全般タブに[ここ](https://www.atmarkit.co.jp/ait/articles/1603/11/news050.html)にあるような「ブロック解除」のチェックボックスはありますか？

agnnn-ag, 2020-01-18:
試してみましたが、特に変化は見られませんでした。

KawaSwitch, 2020-01-18:
この.dllは64bitOS用だったかもしれません
お使いのOS(Windows10)は32bit/64bitどちらですか？

agnnn-ag, 2020-01-18:
64bitのほうです。

agnnn-ag, 2020-01-18:
今、九日前にダウンロードした（2807KBのもの)を展開して重複するものを上書きしたところ、解決してUIが表示されたのですが、これはどういうことなのでしょうね...

KawaSwitch, 2020-01-18:
このdllは接続されているカメラの名前を取得するために使っているのですが, これ（とそれに関連するコード）を追加したのが5日前なのでそれ以前のものは動くのかもしれません
UIのカメラの部分はIDの直接入力/カメラ名選択のどちらになっていますか？

agnnn-ag, 2020-01-18:
なるほど、です。直接入力になっていました。

agnnn-ag, 2020-01-18:
お手を煩わせてしまい、すいませんでした。ご丁寧なアドバイスとても助かりました。開発頑張ってください。

closed 2020-01-18 by agnnn-ag

KawaSwitch, 2020-01-19:
問題の原因の特定にいたらずにすみません
後のバージョンではdll読み込みに失敗したらその機能を使わない（カメラIDを直接入力する）ように変更しようと思います
情報のご提供ありがとうございました