KawaSwitch/Poke-Controller #9: make時に途中で止まってしまう

goriwaku, 2020-01-22:
はじめまして。
Windows10で書いてある通りのソフトウェアをインストールし、Poke-Controller-Masterをmakeしようとしたのですが、
Fatal error: can't create obj/Joystick.o: No such file or directory
とでてしまい、makeが途中で止まってしまいます。
すでに閉じられていたissueに同様の話題がありましたが、objファイルの作成法が良くわからないので教えていただけると幸いです。
よろしくお願いします。

goriwaku, 2020-01-22:
すみません、私の読み違えでobjディレクトリを作ったらそのエラーはなくなりました。
しかし、ほかの警告がたくさん出てmakeは成功しないので対処法をご教授願いたいです。

avr-gcc -c -pipe -gdwarf-2 -g2 -mmcu=atmega16u2 -fshort-enums -fno-inline-small-functions -Wall -fno-strict-aliasing -funsigned-char -funsigned-bitfields -ffunction-sections -I. -DARCH=ARCH_AVR8 -DF_CPU=16000000UL -mrelax -fno-jump-tables -x c -Os -std=gnu99 -Wstrict-prototypes -DUSE_LUFA_CONFIG_HEADER -IConfig/  -I. -I./lufa/LUFA/.. -DARCH=ARCH_AVR8 -DBOARD=BOARD_NONE -DF_USB=16000000UL   -MMD -MP -MF obj/Joystick.d Joystick.c -o obj/Joystick.o
Joystick.c:27: warning: function declaration isn't a prototype
Joystick.c: In function 'SetupHardware':
Joystick.c:66: warning: implicit declaration of function 'clock_prescale_set'
Joystick.c:66: error: 'clock_div_1' undeclared (first use in this function)
Joystick.c:66: error: (Each undeclared identifier is reported only once
Joystick.c:66: error: for each function it appears in.)
Joystick.c: In function 'ParseLine':
Joystick.c:239: warning: format '%hx' expects type 'short unsigned int *', but argument 4 has type 'uint16_t *'
make: *** [obj/Joystick.o] Error 1

お手数おかけしますが、よろしくお願い申し上げます。

goriwaku, 2020-01-22:
何度もすみません、自己解決しました。
power.hの|| defined(__AVR_ATmega16U2__)の両端の__が抜けていただけでした。
ご迷惑おかけしました。
これからもありがたく使わせていただこうと思います。

KawaSwitch, 2020-01-22:
遅れて気づきました。すみません。
#1 と #2 でそれぞれ自己解決されたようですね。

どちらの回答にも画像が無く, #2に関しては両端の_(アンダーバー)を記載していなかったため分かりづらかったかもしれません。
これらに関して先ほど修正させて頂きました。
ご迷惑をおかけして申し訳ありませんが今後ともよろしくお願いいたします。

closed 2020-01-22 by KawaSwitch

goriwaku, 2020-01-22:
私の理解力が乏しかっただけなので気にしないでください！本当にありがとうございます！
別件でもう一つお伺いしたいのですが、キャプボでつないだ時のCamera IDの取得方法がわからないのですが、教えていただけますでしょうか？
たびたびすみません！よろしくお願いいたします。

KawaSwitch, 2020-01-22:
他にカメラデバイスを何も接続していない場合はIDは0になるはずです
他に接続している場合, 順番にずれていくので1, 2, ...と試していってみてください
※ノートPCなどの場合は内蔵カメラにもIDが振られるため1つずれることに注意

goriwaku, 2020-01-22:
順に試したらできました！何度もありがとうございます！

KawaSwitch, 2020-01-22:
了解いたしました。ご報告いただきありがとうございました！