KawaSwitch/Poke-Controller #13: Poke-Controllerの実行ができない

superd241, 2020-02-09:
はじめまして。
5のPoke-Controllerの実行が何度やってもできないのですが教えていただけると幸いです。
必要なパッケージはすべてインストール済みなのはanaconda navigatorで確認してあります。

KawaSwitch, 2020-02-09:
いくつか状況の確認をさせて頂きます

- 使用OS
- Poke-Controllerのバージョン
どのページのどれをDLしたかでも構いません
- 実行時の状況
ウィンドウは出る/ウィンドウも出ないなど
実行時のAnaconda PromptとPoke-Controllerのスクリーンショットもあると助かります
- 何かしらのエラーメッセージは出ているか

以上4点について, ご回答お願いいたします

superd241, 2020-02-09:
初心者で分からないことだらけですいません。
使用OSはwindous10です。
Poke-Controllerのバージョンは1.0beta
Window.pyのファイルを実行するとこの表示が出て先に進むことが出来ません
[image]

[image]

KawaSwitch, 2020-02-09:
Pythonファイルの実行時は先頭に`python`が必要です
`python C:¥Users¥...¥Window.py`
としてみてください

superd241, 2020-02-09:
無事実行してウィンドウが開きました。
お手数かけてすいません。ありがとうございました。

KawaSwitch, 2020-02-09:
いえいえ、無事実行できたようで良かったです！
ご質問ありがとうございました

closed 2020-02-09 by KawaSwitch