KawaSwitch/Poke-Controller #14: マジカル交換自動化について

ripninetails, 2020-02-11:
全く知識がない状態でしたがこれを機にコードを書き始めたものです

https://qiita.com/pokezaresu/items/b86970a716c64ef46370

上記のサイトや、もともとあるコマンドを参考にして、マジカル交換自動化のコードを書こうとしました。

以下のファイルのコードの内、ポケモンを選択する部分の動作が上手くいかず、pythonでどう書けばいいのか調べても全くわからない状況です

既存の自動リリースのコマンドを少し編集し使用しようと思いましたが、既存のコマンドから、ループを脱出し元のループに戻る方法がわからず断念しました。

動作部分をどうすればいいのか、またはすべて書き直してこれが正解というものがありましたら図々しいのは百も承知ですが追加していただきたいです。

知見を広める非常にわかりやすく便利なツールの作成感謝いたします。

[PythonCommand.zip](https://github.com/KawaSwitch/Poke-Controller/files/4185135/PythonCommand.zip)

Moi-poke, 2020-02-11:
あまり動作確認できていませんが、動作部分の処理を察するにこんな感じでしょうか…

```python3
# マジカル交換
class Magical_Trade(PythonCommand):
	def __init__(self, name):
		super(Magical_Trade, self).__init__(name)
		self.maxpcnt = 30 # ポケモン交換したい数
		self.pcnt = 0 # ポケモンの数

	def do(self):
		while True:
			if self.pcnt < self.maxpcnt:
				self.pcnt += 1
			# 開始
				self.press(Button.Y, wait=1.5)
				self.press(Direction.DOWN, wait=1.0)
				self.press(Button.A, wait=3.0)
				# ポケモン選択
				self.pselect()
				self.wait(1.0)
				self.press(Button.A, wait=1.0)
				self.press(Button.A, wait=6.0)
				self.press(Button.A, wait=1.0)
				self.press(Button.A, wait=1.0)
				self.press(Button.A, wait=90)

				# 交換相手検索中

				self.press(Button.Y, wait=60)
				self.press(Button.A, wait=1.0)
				self.press(Button.A, wait=1.0)
				self.press(Button.A, wait=1.0)
				self.press(Button.A, wait=1.0)
				self.press(Button.A, wait=1.0)

			else:
				self.wait(1.0)
				print('done')
				self.finish()

	def pselect(self):
		bcnt = (self.pcnt - 1) % 30 + 1
		x = (bcnt -1) % 6
		# y = (bcnt -1) / 6
		y = (bcnt -1) // 6 #整数除算

		# for i = 0, i < x, i += 1
		for i in range(0, x, 1):
			self.press(Direction.RIGHT, wait=0.5)

		# for i = 0, i < y, i += 1
		for i in range(0, y, 1):
			self.press(Direction.DOWN, wait=0.5)

```
全体をwhileループに入れました。
pythonのforループは基本的には修正したように書きます。0から始まって5回ループするなら
```python3
for i in range(0, 5, 1):
    hogehoge
```
のような感じです。これは一部省略可能で
```python3
for i in range(5):
    hogehoge
```
とも書けます。詳しくはpython3のforループで調べてみてください。

また、pythonではint型を普通に除算するとfloat型になるので、今回のような場合は整数除算か整数部分を取る必要があります。
[image]

関数定義時・if文の条件後・forループの際など、" : " を書かないと行けない箇所が多いので気をつけましょう。

ripninetails, 2020-02-11:
画像認識を用い交換完了したタイミングですぐYを押す処理を加えた上で動作しました
大変有り難うございます
次にコードを書くときの糧にさせていただきます

closed 2020-02-11 by ripninetails