KawaSwitch/Poke-Controller #23: 自動孵化（画像認識）のが途中で終わってしまう。

zakzakza, 2020-02-18:
色違いを孵化させるためにPythonの自動孵化（画像認識）をスタートさせて、パソコンをつけっぱのまま外出して、帰宅後確認したのですが、なぜか「たまごをもっていたんです」というところで止まっておりました。

正直はっきりとした原因はつかめず、ご教示願いたいです。
（放置する場合はなにかPokeConPCの設定とかPCの設定とかする必要がある？）

一応検証して分かっていることは以下の通りです。
・休日にパソコンで作業している時は10時間経っても動作し続けたままだった。
画像認識のコマンドを実行する際はパソコンの電源やディスプレイの電源とか消えないようにする必要がある？
・Show Realtimeにチェックを外した場合は同様のところで止まっていた。
・「カセキ色厳選」でも同様のことが発生し、途中で止まっていた。

KawaSwitch, 2020-02-18:
ご利用ありがとうございます

提示された内容から推測すると, 画像認識のキャプチャ時に正しくデータが読み取れず, エラーで止まっている気がします

> ・休日にパソコンで作業している時は10時間経っても動作し続けたままだった。
画像認識のコマンドを実行する際はパソコンの電源やディスプレイの電源とか消えないようにする必要がある？

はい. 画像認識のコマンドに限らずPythonのコマンドを実行する際は, PCの電源はつけたままの必要があるため（スリープも不可）, PC側での設定が必要です
ディスプレイの電源はPCがスリープ状態にならないなら消しても大丈夫です

数十分など何も操作が無い場合にスリープする機能を使用している場合は無効にする必要がありますが, 現在の設定はどのようになっていますか？
→ 参考 [スリープ機能の切り方(Windows 10の場合)](https://www.atmarkit.co.jp/ait/articles/1803/22/news018.html)

この機能を切っている状態でも同じ現象が起こる場合は, 止まった場面でのPoke-Controllerのスクリーンショット, およびコンソール(pythonを実行した黒い画面)側のスクリーンショットを貼ってくださると助かります

zakzakza, 2020-02-19:
ご連絡ありがとうございます。

記事に記載された「システム無人スリープ」の設定を行って試したところ、変なところで止まらず色違いのポケをゲットできました！

本当にありがとうございます！

windows10の設定から出てくる電源オプションでスリープは無効にしていたはずなのですが、「システム無人スリープ」というものまでやらないと駄目なんですね。
今思い出すとスリープ無効にしていてもパソコンの電源が落ちているときがたまにあったので不思議でした。

ご教示頂いたようにスリープが原因だと思いますが、念のため今週末までちょっと検証して結果を報告させて頂きます。

KawaSwitch, 2020-02-19:
情報提供感謝いたします

zakzakza, 2020-02-24:
お世話になっております。

連絡遅れました。
あの後もやってて、特に問題なく動作できております。

やはりスリープさせないよう設定が必要ですね。
ありがとうございました！

closed 2020-02-24 by zakzakza