KawaSwitch/Poke-Controller #24: コントローラーについて

nanaC666, 2020-02-29:
PCのゲームパッドで操作したくてControllerGUIにR側Stick関連のボタンを実装しようと、GuiAssets.pyのControllerGUIに（※１）を追記して、UnitCommand.pyに（※２）を追記しました。ボタンの実装はうまくできたのですが、RStick関連のどのボタンを押しても、ずっと右スティックが左上方向？にホールド入力されてしまいます。（■＝方向）
ボタンのログでは[<Stick.RIGHT, ■>]と一度出てきます。他ボタンの入力はしっかり反応します。

※１　（■＝方向）
tk.Button(abxy_frame, text='R_■', command=lambda: UnitCommand.R_■().start(ser)).grid(row=3, column=1)

※２　（■＝方向）
class R_■(UnitCommand):
	def __init__(self):
		super(R_■, self).__init__()
	def start(self, ser):
		super(R_■, self).start(ser)
		self.press(Direction.R_■)

KawaSwitch, 2020-02-29:
コードを拝見したところ, おそらく1.0-beta3より前のバージョンかと思われます
Joystick.cの方に問題がありRスティック関連の問題をつい最近修正しました

"HATスイッチの入力に対応"(コミット番号:[3e06109](https://github.com/KawaSwitch/Poke-Controller/commit/3e061098c7158aab5bbd62bf21669bc5ccc0020e))にてRスティックのバグを修正しています
最新版, もしくは[リリースページ](https://github.com/KawaSwitch/Poke-Controller/releases)からv1.0-beta3のソースコードを取得してもらい, その上で実装してもらった結果を確認したいです

こちらではDirection.R_UP, R_DOWN, R_RIGHTにて正常に動作することを確認いたしました
不便をおかけして申し訳ありませんが, よろしくお願いいたします

KawaSwitch, 2020-02-29:
また, Joystick.cの書き換えに伴い, v1.0-beta3を使用するには[リリースページ](https://github.com/KawaSwitch/Poke-Controller/releases)のv1.0-beta3のAssets内にあるJoystick.hexをマイコンに書き込む必要があります

nanaC666, 2020-03-01:
マイコンがatmega16u2ではない為v1.0-beta3をDLしmakefileのMCU=atmega32u4 に書き換え後、joystick.hex作成、マイコンへ書き込みをして同じように実装したら操作できるようになりました。早急な返信感謝します。v1.0-beta3は初期がHAT（十字ボタン)のみなので、L,RStick両方向の追加が必要ですね。

KawaSwitch, 2020-03-01:
動作のご確認ありがとうございます

> v1.0-beta3は初期がHAT（十字ボタン)のみなので、L,RStick両方向の追加が必要ですね。

方向キーをHATで実装しなおしているので, そうなりますね

また何かありましたらご連絡ください

closed 2020-03-01 by KawaSwitch