KawaSwitch/Poke-Controller #27: コントローラの認識がされない

kochatece, 2020-09-20:
Atmega16U2のマイコンにJoystick.hexを書き込んだのですが，スイッチと接続してもコントローラの認識ができず，Poke-Controllerからの入力が受け付けません．
マイコンはAtmega16U2なのですが，Arduino UNO R3ではありません．同じものを買うべきでしょうか？

KawaSwitch, 2020-09-20:
同じものである必要はありません。ATMega16U2であれば動作するはずです。

まずは、以下3点を確認してください。
- [wiki: マイコンをUSBデバイスとして設定する](https://github.com/KawaSwitch/Poke-Controller/wiki/%E3%83%9E%E3%82%A4%E3%82%B3%E3%83%B3%E3%82%92USB%E3%83%87%E3%83%90%E3%82%A4%E3%82%B9%E3%81%A8%E3%81%97%E3%81%A6%E8%A8%AD%E5%AE%9A%E3%81%99%E3%82%8B)において
手順と同じように書き込んでいき、最後の6においてSuccess（+使用メモリの割合）が表示されたか。
注意点としてArduino互換機の中には安価な代替品としてATMega16U2以外のチップを載せている場合がありますが、ここが通ればおそらく問題ないと思います。

- [wiki: PC-マイコン-Switchをシリアル通信で動かす](https://github.com/KawaSwitch/Poke-Controller/wiki/PC-%E3%83%9E%E3%82%A4%E3%82%B3%E3%83%B3-Switch%E3%82%92%E3%82%B7%E3%83%AA%E3%82%A2%E3%83%AB%E9%80%9A%E4%BF%A1%E3%81%A7%E5%8B%95%E3%81%8B%E3%81%99)において
手順のようにCOMポートを設定し、右側のログに"COM Port [number] connected successfully"と表示されたか。

- 上と同ページにおいて
ジャンパワイヤの接続が正しいか。
シリアル変換チップが正しく動作していれば、TXと記載している部分のランプが信号を送ったときに赤く光るのでそこでも確認できます。

以上、3点の確認をお願いします。

kochatece, 2020-09-21:
ありがとうございます！
確認したところ、無事実装できました。

closed 2020-09-21 by KawaSwitch