KawaSwitch/Poke-Controller #28: 画像認識用の画像の入手

kochatece, 2020-09-26:
画像に関して複数質問させていただきます．
・すでに実装されているカセキ厳選などの画像認識を用いるコマンドがあると思いますが，それに使用する画像は自分で入手しますか？
・画像はCaptureボタンで入手したもの，またはそれをトリミングしたものだけ使用できますか？
・画像の画質はどこまで対応していますか？
・画像はどのフォルダに入れればよいですか？

KawaSwitch, 2020-09-26:
>・すでに実装されているカセキ厳選などの画像認識を用いるコマンドがあると思いますが，それに使用する画像は自分で入手しますか？

現在デフォルトで実装している画像認識必須のコマンドには画像の追加は必要有りません。
仕組みとしてはステータス表示時の色違いマークをテンプレートとして検索しています。

ただし、自分でPythonのコマンドを作成する場合には自分で取得してもらう必要があるかと思います。
<br>

>・画像はCaptureボタンで入手したもの，またはそれをトリミングしたものだけ使用できますか？

テンプレートマッチングにおいてはマッチングのためにキャプチャ元画像とサイズスケールを揃える必要があるため、Captureボタンを押して取得した画像(1280*720)、およびそれをトリミングしたものを利用することを想定しています。
実装予定である他の画像処理/認識においても基本はこれを基準とするつもりです。
<br>

>・画像の画質はどこまで対応していますか？

画像認識に使用する画像サイズを1280*720に固定しているため、現在実装している画像認識の範囲では使用できる解像度はこのサイズで固定となります。
<br>

>・画像はどのフォルダに入れればよいですか？

Poke-Controller\SerialController\Template\
現在、入力画像を選んで画像認識を行うのはテンプレートマッチングの手法のみですが、このメソッドを用いる際にはこのディレクトリ下に配置してください。

kochatece, 2020-09-26:
> 返信ありがとうございます．
画像に関していろいろ知ることができました．プログラム初心者ですが，自分でコマンド作成できるように頑張りたいと思います．

closed 2020-09-27 by KawaSwitch