KawaSwitch/Poke-Controller #31: ポートの認識について

pokeaka, 2020-10-08:
poke controllerを開くことができ、ポートの認識まで行きましたがポートを入力してもファイルが見つからないと出て繋げることができません。デバイスマネージャーを確認しましたがそれでもできません。どうかお教え願えませんでしょうか。

使用OS　windows１０
環境　Poke-Controller v1.0-beta3
[image]

KawaSwitch, 2020-10-08:
これはUSBシリアル変換アダプタをPC側と接続した状態でしょうか？
USBシリアル変換アダプタを接続した際にデバイスマネージャに表示される「ポート（COMとLPT）」の情報が必要となります。
見たところArduinoしか接続されていないように見えます。

pokeaka, 2020-10-08:
シリアル変換アダプタをarduinoにつなげてただけだとダメなのですね。シリアル変換アダプタをパソコンに直接繋いだら認識されました！ありがとうございます。
そして再度質問で申し訳ないのですが試しにA連打やキーボードでのコントロールを実行したところ
Writefile failed (PermissionError１３, 'アクセスが拒否されました。',None,5)とエラーが出て
アクセス拒否されてしまう状態ですこれについてもお教えいただけると幸いです

KawaSwitch, 2020-10-08:
・同時に2つPoke-Controllerのプロセスを開いていないか
・COMポートの番号の設定は正しいか（設定後Reloadを押しているか+その際にログにSuccessと表示されたか）
・他のプログラムで同COMポートを利用していないか
以上、3点の確認をお願いします。

いずれも満たしている場合は、左下のSerial Settingsにチェックを入れて何かしら操作をしたときのスクショが欲しいです。

pokeaka, 2020-10-08:
一度すべて消してから再度接続してみたところ無事に接続が確認できました！
本当にありがとうございました。初心者ながら新しいコマンドを作っていこうと思います！

closed 2020-10-08 by KawaSwitch