KawaSwitch/Poke-Controller #33: 既存のコマンドの編集について

kochatece, 2020-10-17:
お世話になっております。

Poke-Controllerに対する直接的な質問ではないのですが，
既存のコマンドを編集しておりますと，インデントエラーでプログラムが動作しないことが多々あります．
全てのインデントをスペースキーで作り直すと無事動作はするのですが，コマンドのpyファイルを初めて編集するときに限ってインデントエラーが起こり，さきほどの修正作業を行わなければなりません．

VSCodeで編集をしているのですが，エラーの原因や解決方法はありますでしょうか？
お手数おかけしますが，ご教示いただければと思います．

KawaSwitch, 2020-10-17:
お疲れ様です。

私が書いているプログラムはすべてのインデントにタブを用いているので, スペースを混ぜたインデントを行うとPython側でエラーとなることが原因と考えられます。本来はPythonではスペース4個でインデントを行うのが正式なようで(今回知りました), ご不便おかけして申し訳ないです。
ただし, このプロジェクトではとりあえずタブインデントで統一していくつもりです。

解決法としては, VSCodeでは下記リンクのようにタブとスペースを一括変換できるため, こちらを用いるのが良いかと思います。
```
1. F1を押下
2. indent と入力し indentationToSpaces もしくは indentationToTabs を選択
3. Enterを押下
```

<参考>
pep8-ja, Docs, Pythonコードのスタイルガイド
https://pep8-ja.readthedocs.io/ja/latest/#id5

Qiita, VS Code でタブをスペースに変換する手順とショートカット
https://qiita.com/KEINOS/items/ece64ec28dc101096bc7

kochatece, 2020-10-17:
返信ありがとうございます．
スペースとタブが混同するとエラーが起こるのは初めて知りました．勉強になります．
タブで統一化承知しました．自分もタブでインデントを構築するようにします．

VSCodeでの改善案ありがとうございます．今後の制作に活用したいと思います．ありがとうございました．

closed 2020-10-17 by KawaSwitch