KawaSwitch/Poke-Controller #34: atmega32u4での通信について

L1ly1, 2020-10-22:
atmega32u4を利用しています。
[このissue](https://github.com/KawaSwitch/Poke-Controller/issues/26)にもありますが、書き込みを行いましたが動作しません。
32u4用にコンパイルを行い、[このサイト](https://gamepad-tester.com/)でコントローラとして認識されているのを確認しました。
Poke-Controller側からも認識されていますが、ボタン入力が行えません。

32u4を持っていないとのことなので解決できるかはわかりませんがご報告まで。

KawaSwitch, 2020-10-22:
ご報告ならびに情報提供ありがとうございます。確認したいことが3点あります。

#### - エラー表示について
Poke-Controller側で入力を行った際に, 右側のログもしくは同時に立ち上がるコンソール(黒い画面)にエラーメッセージ等が表示されているか。

#### - Poke-Controller直下のJoystick.cを用いたか
[Wikiのページ](https://github.com/KawaSwitch/Poke-Controller/wiki/%E3%83%9E%E3%82%A4%E3%82%B3%E3%83%B3%E3%82%92USB%E3%83%87%E3%83%90%E3%82%A4%E3%82%B9%E3%81%A8%E3%81%97%E3%81%A6%E8%A8%AD%E5%AE%9A%E3%81%99%E3%82%8B)で紹介している[Windowsで.hexファイルを作成する](https://note.com/gamewagashi/n/n2ffdc55cd347)ではgit cloneを実行しているが, それを飛ばしPoke-Controller内のMakefileおよびJoystick.cに対してコンパイルを行ったか。

#### - MCU機能の確認
上の操作を行っていた場合, Poke-Controller/Joystick.cの次の1行を変更してコンパイル, 再度書き込みをお願いしたいです。
```
215行目
Proc_State_t proc_state = NONE;
を
Proc_State_t proc_state = MASH_A;
へ変更
```
コンパイル&書き込み後, シリアル変換機をArduinoに繋げずにPC/Switchへ差し込んだ際にA連打を行うようなプログラムです。
この動作はシリアル変換無しの動作なので, まずは上記の確認サイトでこの動作がうまくいっているか確認してほしいです。

以上3点, よろしくお願いします。

L1ly1, 2020-10-25:
お返事ありがとうございます。
確認したところ[Windowsで.hexファイルを作成する](https://note.com/gamewagashi/n/n2ffdc55cd347)でSwitch-Fightstickファイル内のJoystick.cをコンパイルしていました。
Poke-Controller内のJoystick.cをコンパイルしたところ無事に動きました。

お手数おかけしてすみませんでした。
ありがとうございました。

KawaSwitch, 2020-10-25:
ご確認ありがとうございます。無事通信できたようで何よりです。
ファイル名が被っていて分かりにくいのでWiki内の説明を少し変更しておきたいと思います。

closed 2020-10-25 by KawaSwitch