KawaSwitch/Poke-Controller #36: カラーでの画像認識について

YamperJumper, 2020-11-05:
無知な為、初歩的な質問で申し訳ありません。
画像認識にて色違いの判定をカラーでしたいのですが、
self.isContainTemplate(''.png', 0.7,False )
とすればカラーでテンプレートマッチングできるのでしょうか？

KawaSwitch, 2020-11-05:
合っています。
[Wiki: Pythonコマンド_作成How_to](https://github.com/KawaSwitch/Poke-Controller/wiki/Python%E3%82%B3%E3%83%9E%E3%83%B3%E3%83%89_%E4%BD%9C%E6%88%90How_to)の isContainTemplate にある通り, 第3引数をFalseとするとカラーでのマッチングとなります。
（が, 例文の色違い判定にて間違えてTrueを用いていたので修正いたしました。これが混乱の原因だったら申し訳ないです。）

また, 
```
self.isContainTemplate(~, show_value=True)
```
とすることで, マッチング時の類似度を[-1, 1]で表示することが可能ですので第2引数の閾値の調整にお使いください。

YamperJumper, 2020-11-05:
そうなのです、例文がTrueとなっていたので自信が持てませんでした。無知な故ですのでお気になさらないでください。素晴らしいプログラムでとても便利に使わせていただいています。
類似度の表示はとても助かります、参考にさせていただきますね。
ありがとうございました。

closed 2020-11-05 by YamperJumper