KawaSwitch/Poke-Controller #37: Poke-Controllerの導入について

iamgarugaru, 2020-11-06:
度々すみません。
4.マイコンをUSBデバイス化するがどうしてもうまくいきません
PowerShellでdfu-programmerを起動しようとしても反応が無いのですが、どうしたらいいですか？
マイコンはELEGOO Arduino Uno R3 です
[image]
[image]

KawaSwitch, 2020-11-07:
見たことない状況ですね...。
とりあえず下記コマンドを実行したとき, PowerShell上でどのように表示されるかを教えてください。

```
.\dfu-programmer.exe --version
```
```
.\dfu-programmer.exe --targets
```

また, erase後のflash, reset時も同じように何も表示されないでしょうか？

KawaSwitch, 2020-11-07:
念のためPCの再起動後に実行をお願いします。

iamgarugaru, 2020-11-07:
PCの再起動後、全てのコマンド入力してみましたが反応がありません。
自分でももう少し調べてみます
[image]

KawaSwitch, 2020-11-07:
```
echo "a"
```
ものは試しなのですが, これの実行結果はどうなりますかね？

iamgarugaru, 2020-11-07:
とりあえず、進展として伝えときます
dfu-programmer単体で起動しようとすると以下の感じになるので
[image]
dfu-programmer-win-0.7.2\dfu-prog-usb-1.2.2\x86に配置されている
libusb0_x86.dllをlibus0.dllとリネームしてdfu-programmerと同じフォルダ内に配置しました

そして、PowerShellでコマンド入力後、反応はあったのですがマイコンが認識されてないという感じでした。
[image]

デバイスマネージャーを開いて確認もしてみましたが
[image]
atmega16u2としては認識されていないような感じです。
接続手順も間違ってはいないとは思うのですが。。。。
マイコンを買いなおした方がよさそうですか？

KawaSwitch, 2020-11-07:
USBシリアルデバイスはArduino

KawaSwitch, 2020-11-07:
また, [この記事](https://another.maple4ever.net/archives/2380/)のようにドライバを指定してデバイスを認識させるところまでの流れが必要かもしれません。
そもそもショートからDFUモードに入れればの話ですがどうでしょうか？

iamgarugaru, 2020-11-07:
そうですね。
試してみたのですがダメでした。
ショートの解除前後でPCの反応も無いですね...
マイコンを触ったことがないのでどの状態でDFUモードに入ってるのか分からないのでなんとも判断できないです。
個人的には初期不良であってほしいと願っているのですが（笑）
Amazonで購入しましたし、返送もそこまで大変ではないので一度買いなおしてみて
ダメだったらもう一度考えます！

KawaSwitch, 2020-11-22:
私も最初は純正品を使っていましたが, 現在は同じAmazonで購入した"ELEGOO Arduino Uno R3"で動いているので謎ですね...

デバイスマネージャ上ではショート外して"不明なデバイス"もしくは"Atmel USB Devices"として表示されればDFUモードに入っています。
ハードの不良であればまだ良い（良くはないけど）のですがあまりお力になれず申し訳ないです。

closed 2021-07-10 by KawaSwitch