KawaSwitch/Poke-Controller #38: Poke-Controllerの仕様などについて

kochatece, 2020-11-14:
お世話になっております．

またいくつか質問があります．お時間ありましたらご教示いただければと思います．

・Poke-Controllerでは音声を出力させることは可能でしょうか？
音声認識をしたいというわけではないのですが，BGMがあればいいなと思った所存でございます．

・FPSを60に設定することは可能でしょうか？

・画像認識を多用しているとプログラムの性質上，CPUの使用率が上がると思うのですが，CPUの使用を制限することは可能でしょうか？(例えばCPU使用率を50%までに制限する)

・制作したコマンドはPoke-Controllerウインドウ内のcommandに表示がされますが，表示の順番を入れ替えることは可能でしょうか？また，表示方法はどのように決定されるのでしょうか？

以上よろしくお願いいたします．

KawaSwitch, 2020-11-22:
お疲れ様です。いつもありがとうございます。
なかなか時間がとれずに申し訳ないです。

> Poke-Controllerでは音声を出力させることは可能でしょうか？

現在は不可能です。マルチスレッド下で音声と画像を同時キャプチャし再生する必要があるため負荷が今以上にかかり, 恐らく対応は難しいです。

> FPSを60に設定することは可能でしょうか？

現在は不可能です。Python+OpenCVで実装している以上, パフォーマンスがかなり犠牲になっています。
ただ将来的に, 該当部分をC++のライブラリで書き換えて60fpsへ対応するという目標はあります。

> 画像認識を多用しているとプログラムの性質上，CPUの使用率が上がると思うのですが，CPUの使用を制限することは可能でしょうか？(例えばCPU使用率を50%までに制限する)

不可能です。マシンパワーを上げてもらうことでしか解決策はありません。

> 制作したコマンドはPoke-Controllerウインドウ内のcommandに表示がされますが，表示の順番を入れ替えることは可能でしょうか？また，表示方法はどのように決定されるのでしょうか？

SerialController\Commands\PythonCommands(or McuCommands)
を基準とし, このディレクトリ内のファイルから再帰的に.pyファイルを探索し, その中のクラスを読み込んでいます。
探索の順番は,
基準ディレクトリ -> 下層ディレクトリ -> ...
となっているので, もし順番を変えたいのであればファイル名を"1_hoge.py", "2_hoge.py"など順序付けして基準ディレクトリ内に配置する, などで実現できます（反映にはPoke-Controllerの再起動が必要です）。

KawaSwitch, 2020-11-22:
FPSに関してはOpenCVからTk（GUIフレームワーク）用への変換で時間がかかっているようです・・・
少し考えておきます

kochatece, 2020-11-23:
いつも大変お忙しい中，ご回答いただきありがとうございます．
・CPUの使用率について
パソコンのスペックをあげて，より快適な環境を作っていこうと思います．
・コマンドの表示方法について
プログラム名の変更で順番の入れ替えを実現ということですね．大変参考になります．

FPSの件で今回の問い合わせに至った理由としましては，剣盾でポケモンを逃がすプログラムがあると思いますが，それの動作速度をあげようと，ボタンの入力をいじっていたところ，ステータス画面を認識する際にFPSの影響で少しラグが発生しうまく動作しないということがあったため(自分の調整がよくなかったという部分が大きいですが)です．Tkでの表示には結構処理が多いのですね．勉強になります．ありがとうございました．

closed 2020-11-23 by KawaSwitch