KawaSwitch/Poke-Controller #39: 画像認識の範囲指定

inukaizohyo, 2020-11-28:
はじめまして。
Poke-Controllerの画像認識について質問があります。

画像認識の際、認識させる範囲を画面上の指定範囲内に収める（例えば画面の上半分にあるテンプレート画像は認識させて下半分にある同じ画像は認識させない）といったことは可能でしょうか？
もし可能であるとしたら、どのようなプログラムを組めば良いでしょうか？
なにぶんプログラムについては素人同然であるため無知な質問ではあるかも知れませんが、よろしくお願いします。

KawaSwitch, 2021-07-10:
先ほどのコミット 9c70a2a2e69e223b56c0eff65d8006afa14afa3e で対応しました（現在テンプレートマッチングのみ）。
引数として領域を指定し、次のように使用できます。

area: キャプチャ画像をトリミングする領域.  
tmp_area: テンプレート画像をトリミングする領域.  
いずれもリスト形式 [x_begin, x_end, y_begin, y_end] で指定.  

```
self.isContainTemplate('hoge.png', area=[0, 320, 100, 200])
キャプチャ画像の X:0~319, Y:100~199 の領域にテンプレート画像が存在するか判定

self.isContainTemplate('hoge.png', area=[0, 320, 100, 200], tmp_area=[0, 30, 0, 30])
キャプチャ画像の X:0~319, Y:100~199 の領域にテンプレート画像の X:0~29, Y:0~29 の部分が存在するか判定
```

closed 2021-07-10 by KawaSwitch

ShunPoteMobe, 2021-07-10:
お世話になっております。
上記対応が施されたisContainTemplateメソッドの使い方について質問です。

追加された引数area、tmp_areaを指定していない状態で本メソッドをコールした場合、下記ステップにて配列参照外のエラーが出てしまいました。
`		src = src[area[2]:area[3], area[0]:area[1]] if not area else src # trim`
		
`		template = template[tmp_area[2]:tmp_area[3], tmp_area[0]:tmp_area[1]] # trim`
		
■エラー内容
  File "***\Poke-Controller-master\SerialController\Commands\PythonCommandBase.py", line 174, in isContainTemplate
    src = src[area[2]:area[3], area[0]:area[1]] if not area else src # trim
IndexError: list index out of range

本対応後、範囲指定を行わずにメソッドをコールする（キャプチャ画像もテンプレート画像もトリミングしないで使用したい）場合はどのようにコールすればよろしいでしょうか。
現在はひとまず上記処理だけコメントアウトしてしまっています。

お忙しいところお手数ですがご教示頂ければと思います。

Moi-poke, 2021-08-21:
こちらの問題について，ちょっと考えてみました．(時間経っているので参考になるかわかりませんが)

原因
isContainTemplate呼び出し時に`area`または`tmp_area`の引数を与えなかった場合は初期値の空リストを用いてTrimを行おうとするが，空リストにはindex 0, 1, 2, 3が存在しないため`IndexError`となる．

areaのエラーで止まってるので不明ですが，tmp_areaも未指定だと同様のエラーが出る気がします．

解決案
isContainTemplateの実装をいじるのが楽そうです．

- 以下のようにif文で分岐

```python3
if area != []:
    src = src[area[2]:area[3], area[0]:area[1]] if not area else src # trim
```
```python3
if tmp_area != []:
    template = template[tmp_area[2]:tmp_area[3], tmp_area[0]:tmp_area[1]] # trim
```

ifの条件は`if area:`や`if len(area) == 4:`などあると思います．

- 初期値を設定する

areaは(現状)固定値のため関数定義時に初期化可能
tmp_areaは変動するので別手順

```python3
def isContainTemplate(self,
		template_path, threshold=0.7, use_gray=True, show_value=False,
		area=[0, 1280, 0, 720], tmp_area=[]):
```
```python3
template = cv2.imread(TEMPLATE_PATH+template_path, cv2.IMREAD_GRAYSCALE if use_gray else cv2.IMREAD_COLOR)
h, w = template.shape[:2]
if tmp_area == []:
    tmp_area = [0, w, 0, h]
```

---
思想の違いとしては"空リストじゃないなら処理をする" と "空リストなら処理をする"違いです．

あともう一つ，try-except文を使う方法もあると思います．処理的には一番簡単そうです．

ShunPoteMobe, 2021-08-31:
お忙しいところ回答頂き有難うございます。

つまり
`src = src[area[2]:area[3], area[0]:area[1]] if area else src # trim`
とすればよいということですかね。
上記の対応をPythonCommandBase.pyの該当処理部と、同様の形式でtmp_areaを用いる処理に対しても反映したところ、area・tmp_areaを引数指定しなくても問題なく動作することが確認できました。

ただ、今度はarea・tmp_areaの引数指定をした際、areaの指定サイズがtmp_areaよりも大きい場合にエラーになってしまいました。
こちらにつきましてはまだ関数の使い方に誤りがある可能性がありますのでもう少し調査してみて、それでも解決しない場合は別途新たにissueを発行して質問させて頂こうと思います。

有難うございました。