KawaSwitch/Poke-Controller #44: pythonコマンド上でのスクリーンショット取得について

T-shunsaku, 2021-07-08:
お世話になっております。
以下が実現可能か、可能な場合その方法をご教示頂ければと思います。

・PokeController上でキャプチャしている画面情報をPythonコマンドによってスクリーンショットする

■実現したいこと
　PokeController上でPythonコマンドを実行させて自動操作をさせている際、特定条件成立時にその時点のスクリーンショットを自動で取得したいと考えています。
　
　実現したいこととしては上記の通りの為、例えばPokeController上の「Capture」機能がPythonコマンド内で叩けるような形式でも解決できるのですが、そういった機構は実現可能でしょうか。
　
　当初はOpenCVの機能を用いれば実現できるかと考えたのですが、PokeControllerで画面キャプチャ中に実行するとカメラ情報が既に使用されているのでカメラ情報を開くことができませんでした。
　ImageProcPythonCommandクラスがPokeControllerで使用中のカメラを使用していると思われるので実現方法はありそうなのですが、具体的なコードがわかりませんでした。
　
　当方Python、OpenCVは勉強中の為、お手数おかけしますが解決策があればご教示頂ければと思います。

■動作環境
　PokeController：1.0-beta3

KawaSwitch, 2021-07-08:
> 実現したいこととしては上記の通りの為、例えばPokeController上の「Capture」機能がPythonコマンド内で叩けるような形式でも解決できるのですが、そういった機構は実現可能でしょうか。

可能です。この場合、その機構の方が一番手軽に実装できます。

> 当初はOpenCVの機能を用いれば実現できるかと考えたのですが、PokeControllerで画面キャプチャ中に実行するとカメラ情報が既に使用されているのでカメラ情報を開くことができませんでした。

新たにcv2.VedeoCaptureメソッド等でカメラを開こうとしたという解釈で大丈夫でしょうか？
本ソフトでは、毎フレームごとにOpenCVの関数を用いて画像をキャプチャ（Camera.pyにて）しているので、この画像を流用する形で実装できると思います。

具体的な実装としては、先ほどのコミット fa9dbc3eadc0b6d07bf1f42c9fda7c9b6c1080d4  にて変更したものを参考にしてもらうと、ImageProcPythonCommandクラスにCameraクラスのsaveCaptureメソッドをそのまま実行させてあげるメソッドを作成すると、
```
class ImageProcPythonCommand(PythonCommand):
...
def saveCapture(self):
	self.camera.saveCapture()
```
次のようにコマンド内で書けるようになります。
```
class CaptureTest(ImageProcPythonCommand):
	NAME = 'キャプチャテスト'

	def __init__(self, cam):
		super().__init__(cam)

	def do(self):
 		if ... :
			self.saveCapture()
```
-> /SerialController/Captures/にスクショが保存される

良い機能だと感じたので新しく実装させて頂きました。安定版ではありませんが、プルして使用して頂いても大丈夫です。
お役に立てれば幸いです。

ShunPoteMobe, 2021-07-09:
お世話になっております。
迅速な回答と対応をして頂き有難うございます。

> 新たにcv2.VedeoCaptureメソッド等でカメラを開こうとしたという解釈で大丈夫でしょうか？

まさにその通りです。
具体的なコードを書くべきでした、申し訳ございません。

現在安定版で稼働をさせていましたので、最新版で環境作成しなおして試させて頂こうと思います。

お忙しいところご教示頂き有難うございました。

closed 2021-07-10 by KawaSwitch

ShunPoteMobe, 2021-07-10:
無事実現したいことが実現できました。
有難うございました。