KawaSwitch/Poke-Controller #47: Joystick.hax書き込み後のマイコンの挙動について

amane51, 2021-12-01:
お世話になっております。

Joystick.hax書き込み後のマイコンの挙動についての質問なのですが、
マイコン単体でSwitch、およびPCに接続した際に、Homeを連打している状態になっており、
Wikiにて案内いただいている配線を行っても、Poke-controllerからの入力を受け付けていない状況です。
存在しない通信で動いていることから、マイコンの不具合という認識であっていますでしょうか。
もし、コードの書き換えの必要がありましたら、ご教授願います。

以下現在の環境です。
マイコン→ELEGOO Arduino Uno R3
使用バージョン→Poke-Controller v1.0-beta3
Joystick.hax→上記バージョンのAssets内よりダウンロード

KawaSwitch, 2021-12-01:
お疲れ様です。

配布Joystick.hexにてHomeボタンを連打する挙動は、PC側から受け取ったシグナルが正常でないときに起こります。
その挙動を起こす要因がこのソフトに存在するので、マイコンの不具合ではないかもしれません。
[参考: Wiki - シリアルデータの中身について](https://github.com/KawaSwitch/Poke-Controller/wiki/%E3%82%B7%E3%83%AA%E3%82%A2%E3%83%AB%E3%83%87%E3%83%BC%E3%82%BF%E3%81%AE%E4%B8%AD%E8%BA%AB%E3%81%AB%E3%81%A4%E3%81%84%E3%81%A6)

ただ、Poke-Controller v1.0-beta3で配布しているJoystick.hexでは、初期値がNone（=何も押下しない）の状態のはずです。
マイコン単体で接続した際には、シグナルを待ち続けて入力を行わないのが正しい動作となります。
挙動を確認したいので、一度PC接続を解除し、下記手順でSwitchとマイコン単体で繋ぎ直してみてください。

・[Wiki - マイコンをUSBデバイスとして設定する](https://github.com/KawaSwitch/Poke-Controller/wiki/%E3%83%9E%E3%82%A4%E3%82%B3%E3%83%B3%E3%82%92USB%E3%83%87%E3%83%90%E3%82%A4%E3%82%B9%E3%81%A8%E3%81%97%E3%81%A6%E8%A8%AD%E5%AE%9A%E3%81%99%E3%82%8B)を参考に、[Joystick.hex（20.6KB）](https://github.com/KawaSwitch/Poke-Controller/releases/download/v1.0-beta3/Joystick.hex)をdfuモードで書き込む
・各項目で Success など同じ表示がされているか確認する
・6番目で61.46%の書き込みが行われているか（使用マイコンが同じなので等しくなるはず）確認する
・Switchにマイコンを接続し、挙動を確認する

再度この手順を行ってもなおHome連打およびその他挙動が起こる場合はご連絡ください。

amane51, 2021-12-02:
ご確認ありがとうございます。

上記手順の元、再度Joystick.hexを書き込みましたが、マイコンのみをSwitchに接続した際以下の挙動になっておりました。
・数秒Home連打→接続が切れる(Switch左下のコントローラーマークが消える)
PC-USBシリアル-マイコン-Switchの接続をした際も変わらず、Poke-controllerからの入力は受け付けませんでした。

amane51, 2021-12-02:
Arduino Uno Rev3 ATmega328で同様の手順を行ったところ全く問題なく使用することができるようになりました。
マイコンの初期不良だったようです。

お手数をおかけして申し訳ありませんでした。対応ありがとうございました。

closed 2021-12-02 by amane51