KawaSwitch/Poke-Controller #48: アナコンダ上で「指定されたパス」がみつからない

ookoshidesu, 2021-12-21:
はじめまして。
5. Poke-Controller/SerialControllerまで移動について
アナコンダ上において、画像のように「パスが見つからない」と表示されます
ほかの方の質問を参考に
・使用OS　Windows10
・Poke-Controllerのバージョン　Poke-controller-master

何かダウンロードしてないものがあるのでしょうか、よろしくお願いします。
[image] #

KawaSwitch, 2021-12-21:
お疲れ様です。

[Poke-Controllerまでのパス]は、プレースホルダであり実際のパスの代わりです。
例えばデスクトップにPoke-Controllerのディレクトリがある場合は、
C:\Users\\[ユーザ名]\Desktop\Poke-Controller\SerialController
のようなパスを指定します。

エクスプローラでPoke-Controllerを開き、アドレスバーをクリックして絶対パスを見ることもできます。
[Windows 10でファイルのパスを確認する方法](https://faq.nec-lavie.jp/qasearch/1007/app/servlet/relatedqa?QID=020862)（最終アクセス日: 2021/12/21）

ookoshidesu, 2021-12-21:
無事に起動できました！
ろっこく様の記事でもできなかったのでとても助かりました。
ありがとうございました！

closed 2021-12-21 by KawaSwitch